CircuitVerse/cv-frontend-vue #27: Drop JQuery dependency

tachyons, 2022-10-12:
Though we removed jquery ui, we still have jquery in our codebase. which need to be removed and use native DOM methods

disha1202, 2022-10-12:
Hi @tachyons, 
I would like to work on this.

tachyons, 2022-10-12:
@disha1202 You can start working on this, but since jquery is used in several files, PR will be very huge. So I suggest you create PRs with changes per file. Also multiple people can work on this if you cab mention which file/files you want to work on

disha1202, 2022-10-20:
Sure @tachyons, I would like to start with Verilog2CV.js file.

snehalshah8429, 2022-10-20:
@tachyons can I work on this?

snehalshah8429, 2022-10-20:
@tachyons  I would like to contribute on this

snehalshah8429, 2022-10-25:
@disha1202 do you need any help in this project?

PRO-BROS, 2022-10-28:
Very well. Facing the same issue but now resolved on my [site](https://prhocoding.com/)

disha1202, 2022-11-02:
@tachyons, I've removed JQuery dependency in Verilog2CV.js file.
Please take a look.

GarvitSinghal47, 2023-01-18:
@tachyons and @devartstar as there are multiple js files in which there is still jquery present in them I would like to work and help remove jquery from some of the files.

GarvitSinghal47, 2023-01-25:
Hey @tachyons  as i discussed in the slack i am currently working on circuit.js file.

niladrix719, 2023-10-24:
i am working on action.js

antriksh-9, 2023-12-27:
Hey @Arnabdaz @Prerna-0202 I have worked on PR #226.

antriksh-9, 2023-12-27:
I am working on testbench.js file.

antriksh-9, 2023-12-30:
I am working on tutorials.js file.

antriksh-9, 2024-01-04:
I am working on ux.js file.

antriksh-9, 2024-01-04:
I am working on canvasApi.js file.

tachyons, 2024-01-09:
Note: when you move transitions like fade from JQuery, use [Vue transitions](https://vuejs.org/guide/built-ins/transition) instead of trying to implement it vanilla JS.

closed 2024-08-11 by Arnabdaz

tachyons, 2026-10-13:
Sub-task: #1412 removes jQuery from the ApplyThemes and CombinationalAnalysis Vue dialogs.

PR #29 (2022-11-06): refactor: removed JQuery dependency in tutorials.js file
Fixes #27
PR #28 (2022-11-08): refactor: removed JQuery dependency in Verilog2CV.js file
Fixes #27
PR #34 (2022-12-30): refactor: removed JQuery dependency from verilog.js file
Fixes: #27 

Removed JQuery dependency from verilog.js file
PR #35 (2022-12-30): refactor: removed JQuery dependency from restrictedElementDiv.js file
Fixes: #27 

Removed JQuery dependency from restrictedElementDiv.js file