CircuitVerse/cv-frontend-vue #41: Update the readme with setup guide

disha1202, 2023-01-02:
Update the readme with guide to setup the project on local machine.

Amanshenoy008, 2023-01-16:
Hey, i would like to work on this...

disha1202, 2023-01-16:
> Hey, i would like to work on this...

@Amanshenoy008 
Sure go ahead

Amanshenoy008, 2023-01-19:
What's the commit message format ....

disha1202, 2023-01-19:
> What's the commit message format

Please refer to this https://www.conventionalcommits.org/en/v1.0.0/

Avi-88, 2023-02-03:
Hi @disha1202 , seems like there was already a open PR for this issue. I am extremely sorry @Amanshenoy008 I didn't notice it as it wasn't  linked with the issue. If you were still working on this please let me know I'll close mine, thanks and once again sorry for not noticing before creating the PR

Amanshenoy008, 2023-02-03:
oh well , itz fine .

disha1202, 2023-10-12:
Closing as duplicate of #190

closed 2023-10-12 by disha1202

tachyons, 2023-10-12:
Thanks for raising this issue, I missed this issue when created new one

disha1202, 2023-10-12:
> Thanks for raising this issue, I missed this issue when created new one

No issue