CircuitVerse/cv-frontend-vue #55: Creating New Circuit dialog box - default name should be selected

devartstar, 2023-01-07:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to tabsbar
2. Click on `+` icon
3. New Circuit Dialog Box appears
4. instead of placeholder -> default circuit name should be selected

**Expected behavior**
compare with [circuitverse site](www.circuitverse.org)

**Screenshots**

https://user-images.githubusercontent.com/61665451/211151728-9f351ec4-4e1d-4169-84a7-2dc3e7ddb811.mov

**Device Information [optional]:**
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

**Are you working on this issue? (Yes/No)**
No

ashutosh887, 2023-02-25:
@devartstar Is this Issue resolved? I would like to contribute!

sushilpandeyy, 2023-03-12:
@devartstar I want to contribute in this issue please assign it to me

ashutosh887, 2023-03-12:
@sushilpandeyy I don't think this is active Repository

Adarsh9306, 2023-04-02:
HELLO I WANT TO CONTRIBUTE IF THIS IS STILL OPEN PLEASE ASSIGN IT TO ME

Arnabdaz, 2024-01-03:
fixed #170

closed 2024-01-03 by Arnabdaz