CircuitVerse/cv-frontend-vue #61: Creation of a new circuit while having no circuits cause an error

palashb01, 2023-01-09:
**Describe the bug**
- If a user attempts to delete the only circuit in the system using the `delete circuit` button within the properties panel then,  creating a new circuit causes an error message in the frontend.

**Demo:**
[image]

**Expected behavior**
- the `delete button` should give a warning, if there is only one circuit.

**Device Information [optional]:**
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Are you working on this issue? (Yes/No)**
**YES**

palashb01, 2023-01-18:
> hello @palashb01 , I would like to work on this issue
>  

Sure go ahead 👍

chetanjangir0, 2023-01-19:
@palashb01 Issue resolved, Please review

Arnabdaz, 2024-01-03:
fixed #170

closed 2024-01-03 by Arnabdaz