CircuitVerse/cv-frontend-vue #62: Consistency in new circuit prompts.

palashb01, 2023-01-10:
**Describe the bug**

- There are two different options to create a new circuit.

```
1. Click on the plus sign present inside the tabs bar.
2. Click on the new circuit present inside the drop-down of circuit.
```

- Both the methods do the same thing, however, they have different kind of prompts/ dialogue box.

**Expected behavior**

- Both techniques should use the same dialogue box to keep the UI consistent.

### Screenshots

**Difference:**
[image]

[image]

**Expected:**

[image]

**Device Information [optional]:**
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Are you working on this issue? (Yes/No)**
**YES**

hussain-shariff, 2023-01-16:
@palashb01 I would like to work on this issue.

palashb01, 2023-01-16:
> @palashb01 I would like to work on this issue.

Sure go ahead 👍

hussain-shariff, 2023-01-16:
@palashb01 thanks

Avi-88, 2023-02-16:
Hi @hussain-shariff , are you still working on this issue ?

hussain-shariff, 2023-02-18:
@Avi-88 nope, go ahead

Avi-88, 2023-02-19:
Thanks, I'll start with making the PR then

Vasant18, 2023-04-03:
hey **@Avi-88**, are you still working on the issue?

akoolarni, 2023-11-04:
Can I take the issue if not assigned to anyone?