CircuitVerse/cv-frontend-vue #68: Error while creating a new circuit after deleting

GarvitSinghal47, 2023-01-27:
**Describe the bug**
when we try to add a new circuit after deleting any circuit from the delete circuit button then it show error as displayed below.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'circuitverse simulator'
2. Click on 'plus icon to generate some new circuit and create some circuit'.
3. Then go to any circuit and delete it using delete circuit button .
4. Then using plus icon try to add some new circuit in it .
5. See error

**Screenshots**
[image]

**Are you working on this issue?  No**

jaydip1235, 2023-11-18:
@vedant-jain03 @Arnabdaz @Prerna-0202 , Can I work on this issue ?

Arnabdaz, 2024-01-03:
fixed by #170

closed 2024-01-03 by Arnabdaz