CircuitVerse/cv-frontend-vue #71: ui: Quick-Button is overlapping the Navigation bar

niladrix719, 2023-02-01:
**Describe the bug**
Quick-Button is overlapping the Navigation bar

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://circuitverse.netlify.app/simulator'
2. Decrease the screen to enable the hamburger icon
3. Click on 'hamburger icon'
4. See error 'The quick button is overlapping'

**Expected behavior**
The Quick-button should not overlap the Navigation bar

**Screenshots**

[image]

- OS: [Ubuntu 22]
- Browser [chrome]
- Version [v22]

**Are you working on this issue? (Yes/No)**
Yes

Mayank1170, 2023-02-09:
Hey @niladrix719 !!
I'll already start working on this issue and actually capture the issue in the code..
so can you please assign it to me ?

Mayank1170, 2023-02-10:
Hey @niladrix719 
Am i doing right ?

[Screencast from 2023-02-10 21-30-35.webm](https://user-images.githubusercontent.com/98695348/218137836-4054f05b-76ee-46f0-8d85-ca28d0cb0a6f.webm)

DhairyaMajmudar, 2024-01-06:
@niladrix719 shall I take up this?
Thank you.

closed 2024-06-02 by Arnabdaz