CircuitVerse/cv-frontend-vue #80: Disable text underline on hovering items in Nav Menu

devartstar, 2023-02-27:


rockharshitmaurya, 2023-02-28:
I'm reaching out to request that you could assign me this particular issue, I'm interested in working on. Specifically, I'm interested in tackling issue #80. I believe this issue aligns with my skills and experience, and I'm excited to contribute to the project in this way.

I would appreciate it if you could assign me this issue and provide any guidance or suggestions for getting started.

Thank you for your time and consideration.

kr-2003, 2023-03-01:
I want to work on this issue! Can you assign this to me?

closed 2023-03-03 by devartstar

GaganMeshram123, 2023-12-24:
hey @Prerna-0202 and @vedant-jain03  Is these issue open to contribute ?
i would like to contribute on these issue.

Arnabdaz, 2023-12-25:
This issue is already closed as completed

PR #87 (2023-03-03): fix: disabled underline on hovering items in Nav Menu
Fixes #80 

#### Describe the changes you have made in this PR -
I just add "text-decoration:none" under ".navbar-menu > li > a:hover" in "src/styles/css/main.stylesheet.css" file.