CircuitVerse/cv-frontend-vue #81: Padding and Alignment issue in Tabsbar

devartstar, 2023-02-27:
**Describe the bug**

Things to fix -

- Tabsbar height
- cross icon
- alignment of circuit name with cross icon
- plus icon size and alignment

**To Reproduce**

Steps to reproduce the behaviour:

- Go to [cv-frontend-vue](https://circuitverse.netlify.app/)
- check for the styling issue in tabsbar
- compare the difference with [current simulator](https://circuitverse.org/simulator)

[image]

JayashTripathy, 2023-02-28:
I want to work on this.

rockharshitmaurya, 2023-02-28:
Hey @devartstar I would love to work on this issue, could you please assign me this issue

devartstar, 2023-03-01:
Hi @rockharshitmaurya since @JayashTripathy asked first, let him go forward with this. 
Will open more issues soon, you can pick them up.

Rajeevjewar, 2023-05-25:
Is this issue still opens, If it is so please assign it to me. I want to work on this issue.

harshsharma-11, 2023-08-19:
@devartstar is this still open please let me know i want to work on it

souravgope, 2023-11-25:
@devartstar i want to work on this issue please assign me

krishnagitcontri, 2023-12-19:
Hey @devartstar is the issue still open ?

uddeshyatyagi123, 2023-12-25:
@devartstar , please assign me, I want to work on this issue.

ShivamPande18, 2023-12-29:
hey @devartstar if it is open, I would like to work on this

closed 2024-01-03 by Arnabdaz