CircuitVerse/cv-frontend-vue #82: Custom Theme Dialog box styling

devartstar, 2023-02-27:
**Describe the bug**
correct the ui of the `custom theme` dialog box.

**To Reproduce**
Steps to reproduce the behaviour:

- Go to [cv-frontend-vue](https://circuitverse.netlify.app/)
- From Navbar hover on Tools
- Select Themes from dropdown
- Select Custom theme
- compare the difference with [current simulator](https://circuitverse.org/simulator)

current -

[image]

expected -

[image]

thisiskaransgit, 2023-02-28:
Hey @devartstar, I would like to work on this one!

devartstar, 2023-02-28:
@thisiskaransgit sure go ahead.

kr-2003, 2023-03-03:
Hey @devartstar, is this issue still open. Can i work on this?

PR #97 (2023-03-04): style: custom theme dialog box styling
Fixes #82

#### Describe the changes you have made in this PR -
Changed styling of custom theme dialog box and made it responsive.

### Screenshots of the changes (If any) -

[image]

Note: Please check **Allow edits from maintainers.** if you would like us to assist in the PR.
closed 2023-03-04 by devartstar