CircuitVerse/cv-frontend-vue #83: Toggling in and out of Preview Mode

devartstar, 2023-02-27:
**Describe the bug**
Currently on going to `Preview mode`, the quick button does not disappear and the `Exit Preview button` does not appear. 
To exit you can use `Esc button`.

**To Reproduce**
Steps to reproduce the behaviour:

- Go to [cv-frontend-vue](https://circuitverse.netlify.app/)
- From quick button panel
- Select preview option
- compare the difference with [current simulator](https://circuitverse.org/simulator)

[image]

GarvitSinghal47, 2023-02-28:
@devartstar i have already solve this issue and raised a pr previous month regarding this , have a look at it , i think my pr will resolve the issue.

devartstar, 2023-02-28:
@GarvitSinghal47  Thanks for notifying!

closed 2023-02-28 by devartstar