CircuitVerse/cv-frontend-vue #94: Pop-up does not close when clicking on the close button after clicking 'Report an Issue' button on home page

rockharshitmaurya, 2023-03-03:
### Describe the bug
When clicking on the close button in the pop-up that appears after clicking the "Report an Issue" button on the home page, the pop-up does not close.
### To Reproduce
Steps to reproduce the behavior:
1. Go to the home page.
2. Click on the "Report an Issue" button in the bottom right corner.
3. Wait for the pop-up to appear.
4. Click on the close button in the top right corner of the pop-up.
5. Notice that the pop-up does not close
### Expected behavior
When clicking on the close button in the pop-up, it should close and disappear from the screen.
### Video :

https://user-images.githubusercontent.com/46915044/222689971-0100b894-d7f1-4898-bfca-a4d30ab97bc0.mov

### Are you working on this issue? (Yes/No)
Yes

disha1202, 2023-03-03:
Hi @rockharshitmaurya, 
This issue has already been identified in #36  and also fixed in #37

rockharshitmaurya, 2023-03-03:
Hi @rockharshitmaurya thanks for letting me know , I'm closing the issue

closed 2023-03-03 by rockharshitmaurya