CircuitVerse/cv-frontend-vue #99: Navbar is not responsive in the simulator section.

Mayank1170, 2023-03-03:
**Describe the bug**
Navbar hide behind Properties and Circuit  Elements panel when screen size is  996 x 948 or less than this.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://circuitverse.org/
2. Click on "Simulator" on the navbar
3. Inspect the page and change the screen size to 996 x 948.
4. Click on the navbar-toggler-icon 
5. See error

**Current behavior**
[image]

**Expected behavior**
[image]

**Device Information:**
- Fedora
- Firefox

**Are you working on this issue? (Yes/No)**

neerajbachani, 2023-03-04:
can i work on this?

nisha-laur, 2023-10-26:
Hi, if this issue is still open . I would like to work on this . If you could assign this to me .

souravgope, 2023-11-28:
i want to work on this issue please assign me

codebybishwa, 2024-01-11:
I think the navbar is not responsive for the mentioned size. I intend to align it in a slight portrait view for a better look.
I would like to work on this issue. Could you please assign this issue to me?