CircuitVerse/cv-frontend-vue #113: Add Bengali language translation i18n

Arnabdaz, 2023-03-07:
**Is your feature request related to a problem? Please describe.**
Currently there is only English and Hindi version of i18n available for the simulator

**Describe the solution you'd like**
Also add support for Bengali with i18n

**Describe alternatives you've considered**
Suggestions are welcome

**Additional context**
[image]

**Are you working on this? (Yes)**

DhairyaMajmudar, 2024-01-03:
@Arnabdaz I also want to add Marathi and Gujarati language shall I go ahead?

Murdock9803, 2024-01-04:
@Arnabdaz @DhairyaMajmudar I like the idea of more languages support by Dhairya. I also want to add Telugu and Malayalam languages support. This will help students from south india. So shall I go ahead ?

Arnabdaz, 2024-01-04:
i think we would be having something similar to https://www.transifex.com/ we can discuss regarding the same on circuitverse slack channel

PR #114 (2024-07-28): feat(i18n): upadate i18n, add bengali support for i18n
Fixes #113

#### Describe the changes you have made in this PR -
Added i18n Bengali support

### Screenshots of the changes (If any) -
[image]

Note: Please check **Allow edits from maintainers.** if you would like us to assist in the PR.
closed 2024-07-28 by JoshVarga