CircuitVerse/cv-frontend-vue #115: fix: add gap between each timing-diagram

kr-2003, 2023-03-08:
**Describe the bug**
Add spacing between each timing-diagram under under timing-diagram modal.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to [https://circuitverse.netlify.app/simulator](https://circuitverse.netlify.app/simulator)
2. Click on Timing Diagram
3. See error
[image]

**Expected behavior**
[image]

**Are you working on this issue? Yes**

AbdulBasit2733, 2023-04-13:
can i work on this issue.

nisha-laur, 2023-10-26:
Hi, if this issue is still open . I would like to work on this . If you could assign this to me .

disha1202, 2023-10-26:
Hi @nisha-laur, 
This issue has been fixed.

closed 2024-01-03 by Arnabdaz