CircuitVerse/cv-frontend-vue #132: tabsbar section is behaving not optimal when we add multiple tabs

JayashTripathy, 2023-03-14:
**Describe the bug**
tabs are behaving not optimal  , it should overflow and scroll when we add multiple tabs but it makes a new column

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '/simulator'
2. Add multiple tabs (around 8-10)
3. See error

**Current behavior**

https://user-images.githubusercontent.com/76092296/225118741-a9b7990e-823c-4835-a5b3-0f5470ab2065.mp4

**Expected behavior**

https://user-images.githubusercontent.com/76092296/225118623-505d9209-77c6-4a91-9388-c1b3d5fbd2eb.mp4

**Screenshots**
If applicable, add screenshots to help explain your problem.

- Browser [e.g. chrome, firefox]

**Are you working on this issue? (Yes/No)**

Yes

JayashTripathy, 2023-03-14:
hey @devartstar can you please review it.

Mayank1170, 2023-03-19:
Hey @JayashTripathy can you please assign this issue to me !