CircuitVerse/cv-frontend-vue #155: Cross/Close button not working in "Report an issue" section on simulator page

Mitali2700, 2023-03-31:
**Describe the bug**
The cross/close button in the "Report an issue" section of the CircuitVerse frontend simulator page is not working. Clicking on it does not close the window.

**To Reproduce**
Steps to reproduce the behavior:
Go to the CircuitVerse frontend simulator page.
Click on the "Report an issue" button.
Fill out the form and click the "Submit" button.
Try to close the window using the cross/close button in the top right corner.
Notice that the window does not close.
Expected behavior
When clicked, the cross/close button should close the "Report an issue" window.

**Screenshots**
<br>
[image]
<br>

[image]
<br>
after clicking on **report button** it shows:
[image]

**Device Information:**
OS: Windows 11
Browser: Chrome

**Are you working on this issue? **
Yes

Adarsh9306, 2023-04-02:
Hello @Mitali2700 if this issue is open can you please assign it to me? ... I am interested to work on this

ishsarin, 2023-07-31:
Hi, if this is open can I contribute?

Arnabdaz, 2023-08-03:
fixed by pr #174

closed 2024-01-03 by Arnabdaz