CircuitVerse/cv-frontend-vue #164: At the zoom section the zoom in and zoom out button are not in order they are little up and down which look bad

Nitishupkr, 2023-05-16:
**Describe the bug**
At the zoom section the positive and negative button are not in order they are little up and down which look very bad we can align them in proper odder providing proper margin to it

**To Reproduce**
Steps to reproduce the behavior:
1. Go to  'https://circuitverse.netlify.app/simulator'
[image]

**Expected behavior**
[image]

**Device Information [optional]:**
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

**Are you working on this issue? (Yes/No)**
Yes

Nitishupkr, 2023-05-16:
@devartstar see it

Kuroo7, 2023-06-07:
Can you assign it to me

ishsarin, 2023-07-31:
Hi, if this is open can I contribute?

vatan45, 2023-10-02:
hi, if this is open i can raise a pr as i am on a verge of removing this issue

nisha-laur, 2023-10-26:
Hi, if this issue is still open . I would like to work on this . If you could assign this to me .

closed 2024-01-03 by Arnabdaz