CircuitVerse/cv-frontend-vue #188: Deprecation Warning for Division Operator in Sass

YashGupt29, 2023-10-02:
Issue Summary:

I encountered a deprecation warning while using FontAwesome that's related to the use of the division operator (/) in Sass. This warning indicates that starting from Dart Sass version 2.0.0, using / for division outside of a calc() function is deprecated.

Recommendation:

The recommendation provided in the warning message is to use e
ither the math.div() function or calc() to perform division in Sass code.

Impact:

This deprecation warning affects the compatibility of FontAwesome with future versions of Dart Sass, potentially causing issues for users who rely on FontAwesome in their projects.

Steps to Reproduce:

Install FontAwesome and use it in a project.
Compile Sass code that uses the division operator (/) for division outside of a calc() function.
Observe the deprecation warning.

Are you working on this issue?
yes

[image]

YashGupt29, 2023-10-02:
@Arnabdaz  sir can u please review it! if possible assign me this issue!

Arnabdaz, 2023-10-03:
> @Arnabdaz sir can u please review it! if possible assign me this issue!

sure you can work on this issue go ahead

Arnabdaz, 2024-01-03:
fixed by #214

closed 2024-01-03 by Arnabdaz