CircuitVerse/cv-frontend-vue #191: Hover animation not as expected

DhairyaMajmudar, 2023-10-12:
**Describe the bug**
while hovering on the tabs present in the navbar its showing improper animation it must be corrected

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://circuitverse.netlify.app/simulatorvue
2. Hover on navbar tabs
3. See error

**Expected behavior**
The hover effect must be a bit better

**Screenshots**
[image]

**Device Information [optional]:**
- OS: windows
- Browser FireFox
- Version Windows 11

**Additional context**
Add any other context about the problem here.

**Are you working on this issue? (Yes/No)**
YES

DhairyaMajmudar, 2023-10-12:
@Prerna-0202 pls. review and assign this issue to me.

Prerna-0202, 2023-10-13:
> @Prerna-0202 pls. review and assign this issue to me.

Could you please in a bit more detail, like how you want to change?

DhairyaMajmudar, 2023-10-13:
> > @Prerna-0202 pls. review and assign this issue to me.
> 
> Could you please in a bit more detail, like how you want to change?

On hover the arrow svg get shifted in right side and also the underline is not showing a proper transition while hovering .

nisha-laur, 2023-10-26:
Hi, if this issue is still open . I would like to work on this . If you could assign this to me .

akoolarni, 2023-11-04:
I am ready to take the issue. Can I?

pankaj-2503, 2023-12-31:
I don't think there modification is required as the hover effect is fine as on hover it creates a bar under links and removes the angular drop down. No need of any changes. Close this issue.

closed 2024-02-15 by DhairyaMajmudar