CircuitVerse/cv-frontend-vue #194: Navbar hover dropdown is going behind Circuit Elements

jaydip1235, 2023-11-18:
**Describe the bug**
 Navbar hover dropdown is going behind Circuit Elements on mimimizing circuit elements and then maximizing it.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://circuitverse.netlify.app/simulatorvue'
2. Minimize the Circuit elements using minus button
3. Again expand the Circuit Elements
4. Hover over the Project tab and notice that the dropdown is going behind, and not all options are visible even if you minimize the circuit element

**Expected behavior**
 The dropdown elements should come in front of Circuit Elements

**Screenshots**
[image]

https://github.com/CircuitVerse/cv-frontend-vue/assets/65560026/8a03e8c7-9dda-450e-9474-0e52bc22018c

**Device Information [optional]:**
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

**Are you working on this issue? (Yes/No)**
Yes

jaydip1235, 2023-11-18:
@vedant-jain03 @Prerna-0202 @Arnabdaz , Please assign this issue to me

Prerna-0202, 2023-11-19:
Go ahead @jaydip1235

PR #201 (2023-12-18): feat(drag): bug fixes for drag.ts
Fixes #194 

#### Describe the changes you have made in this PR -
-Changes in Drag.ts, changed the z-index.

### Screenshots of the changes (If any) -
Before changes: -
[image]
After changes: -
[image]

Note: Please check **Allow edits from maintainers.** if you would like us to assist in the PR.
closed 2023-12-18 by vedant-jain03