CircuitVerse/cv-frontend-vue #195: Draggable components are overlapping with the navbar.

ArnabBCA, 2023-11-21:
**Describe the bug**
The draggable components are overlapping with the navbar

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://circuitverse.org/simulator
2. Drag the components

**Expected behavior**
The draggable components should be draggable only inside the canvas area.

**Screenshots**
[image]

**Are you working on this issue? (Yes)**

ArnabBCA, 2023-11-21:
I would like to work on this issue.

tanmoysrt, 2023-11-21:
cc @Prerna-0202 @Arnabdaz

ArnabBCA, 2023-11-21:
> cc @Prerna-0202 @Arnabdaz 

May I know what you are doing?

tanmoysrt, 2023-11-21:
> > cc @Prerna-0202 @Arnabdaz
> 
> May I know what you are doing?

They can verify the issue and assign you !

Prerna-0202, 2023-11-22:
Hey @ArnabBCA, Thank you for opening an issue.
I am just curious to know. Is it a suggestion, or do you believe it is a bug? If you consider it a bug, I would like to understand how it could be beneficial.

ArnabBCA, 2023-11-22:
> Hey @ArnabBCA, Thank you for opening an issue.
> I am just curious to know. Is it a suggestion, or do you believe it is a bug? If you consider it a bug, I would like to understand how it could be beneficial.
> 
> 
> 
> 
> 
> 
> 

I think it is a bug because the draggable components should only be draggable inside the Canvas area. Currently it is restricted inside the whole body. As a result it is not considering the navbar and tabs bar height.

ArnabBCA, 2023-11-22:
> I think it is a bug because the draggable components should only be draggable inside the Canvas area. Currently it is restricted inside the whole body. As a result it is not considering the navbar [...]

If it is overlapping the navbar the user cannot do any navbar actions. He/she  has to drag again inside the Canvas in order to use the navbar actions

ArnabBCA, 2023-11-23:
@Prerna-0202 can I proceed?

Prerna-0202, 2023-11-23:
I understand your perspective, but the current setup aligns with the expected behavior. You can check out other open issues.

nickhil-verma, 2023-11-28:
Is this issue Open and should i work on it ??

Arnabdaz, 2023-12-22:
This issue is closed as it is the intended behavior.

closed 2024-01-03 by Arnabdaz