CircuitVerse/cv-frontend-vue #196: Several Console Errors.

DhairyaMajmudar, 2023-12-06:
**Describe the bug**
While running the app in the local server several console errors / warnings are shown in the console.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the app locally using the command `npm run dev`
2. Open the console and check the errors / warnings 

**Expected behavior**
The console must be free from warnings

**Screenshots**
[image]

**Device Information [optional]:**
- OS: Linux - Debain 
- Browser : Firefox
- Version 22

**Additional context**
Add any other context about the problem here.

**Are you working on this issue? (Yes/No)**
Yes

DhairyaMajmudar, 2023-12-06:
@tanmoysrt @Prerna-0202 Several console errors/warnings are noted this might be because of the old version of the font-awesome lib. ( **5.14.4** ) used in the project. 

PTAL and assign this issue to me. 

Thanks!

disha1202, 2023-12-06:
Hi @DhairyaMajmudar 

There's already an existing issue for these warnings #188

DhairyaMajmudar, 2023-12-06:
> Hi @DhairyaMajmudar
> 
> There's already an existing issue for these warnings #188

Thanks,  Shall I open a PR to solve this issue

PR #214 (2024-01-03): Removing console warnings and errors
Fixes #196 

#### Describe the changes you have made in this PR -

 - Upgraded the font awesome from V5 to V6 to remove the console errors / warnings 
 - Commented the hamburger.css file import causing warnings in the console 
 - commented `import { defineProps, defineEmits } from 'vue' ` since its not required any more.  
 - Removed JQuery from `color_theme.scss` 
 - Removed js comments from CSS files 
 - Removed warning from console while building the simulator  

### Screenshots of the changes (If any) -

[image]

Note: Please check **Allow edits from ma [... cut]
closed 2024-01-03 by Arnabdaz