CircuitVerse/cv-frontend-vue #199: Website not matching main repo

Yarlord, 2023-12-09:
**Describe the bug**
The website https://circuitverse.org/simulator seems to be running on a different version than the main branch on this repo. I'm working on a PR to fix issue #198. I don't know what changes to make since it might lead to merge conflict. When I compared this to the currently running website, it had some changes. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to [Current Website](https://circuitverse.org/simulator)
2. Click on Combinational Analysis.
3. Generate a table.
4. Now run the main branch of this repo locally.
5. Go to Combinational Analysis.
6. Generate table.

**Expected behavior**
The UI for both, this repo and the current website were expected to be the same, but there were some changes. 

**Screenshots**
[image]
[image]

**Device Information [optional]:**
- OS: Windows
- Browser: Brave
- Version: 1.6.101

**Are you working on this issue? (Yes/No)**
Yes.

Arnabdaz, 2023-12-22:
@Yarlord the simulator at [https://circuitverse.org/simulator](https://circuitverse.org/simulator) is the old simulator whose code is in the main repo you can check the vue simulator at [https://circuitverse.netlify.app/simulatorvue](https://circuitverse.netlify.app/simulatorvue)

Yarlord, 2023-12-26:
> @Yarlord the simulator at https://circuitverse.org/simulator is the old simulator whose code is in the main repo you can check the vue simulator at https://circuitverse.netlify.app/simulatorvue

Understood, thank you!

closed 2023-12-26 by Yarlord