CircuitVerse/cv-frontend-vue #202: In Render image container the " current view button" is too big for mobile view

upsaurav12, 2023-12-13:
**Describe the bug**
In render image box the "current view button " the button is too big when we see it on the mobile view especially when we change along height as well 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://circuitverse.org/simulator
2. Click on "Download as image button "
3. See the "Current view button " in mobile view
[image]

**Expected behavior**
The button should not big in mobile view
[image]

**Screenshots**
[image]

**Device Information [optional]:**
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

**Are you working on this issue? Yes

upsaurav12, 2023-12-18:
I want to work on this issue

Arnabdaz, 2023-12-22:
@upsaurav12 this issue is related to the old simulator you can open issue for the same at [here](https://github.com/CircuitVerse/CircuitVerse) and close this issue.

upsaurav12, 2023-12-22:
closing this issue

closed 2023-12-22 by upsaurav12