CircuitVerse/cv-frontend-vue #203: Context-menu id  is not responsive not displaying properly

upsaurav12, 2023-12-14:
**Describe the bug**
When you click right click on mouse inside the simulation area then is not responsive as we reduces the width 
but when i checked this when i clicked towards left in the whiteboard then it working fine
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '.https://circuitverse.org/simulator'
2. Right Click on 'white-board '
3. Reduce the Width 
4. See error

**Expected behavior**
It should display properly 

**Screenshots**
I don't know its sometime working sometime not 

https://github.com/CircuitVerse/cv-frontend-vue/assets/116784047/641d6ffc-da79-44da-b8ab-b87587b62aa3

**Device Information [optional]:**
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

**Are you working on this issue? Yes

rose2221, 2023-12-14:
@upsaurav12  i would like to work on this issue. i've already started working on it, i'll upload a draft PR for the same soon. Could you please assign me this issue?

upsaurav12, 2023-12-14:
i am not the maintainer

Arnabdaz, 2023-12-22:
@upsaurav12 this issue is related to the old simulator you can open issue for the same at [here](https://github.com/CircuitVerse/CircuitVerse) and close this issue.

closed 2023-12-22 by upsaurav12