CircuitVerse/cv-frontend-vue #209: Resp: Make mobile landscape view of the simulator.

DhairyaMajmudar, 2023-12-17:
**Is your feature request related to a problem? Please describe.**
Currently, the mobile view of the simulator is not as expected various components are overlapped on each other.

**Describe the solution you'd like**
To make buttons for each and every draggable component which on click opens that tab.

**Describe alternatives you've considered**
N/A

**Additional context**

Expected

Current 

**Are you working on this? (Yes/No)**
Yes

DhairyaMajmudar, 2023-12-17:
@Arnabdaz @Prerna-0202 for doing this I will be opening individual PRs for each button components 
Would be happy to get a better approach from your side. 

Thank You : )

DhairyaMajmudar, 2023-12-25:
@Arnabdaz Just a ping for PR review 
Thank You : )

Arnabdaz, 2023-12-25:
as these are subtask of possible task of a possible gsoc project we might not merge till next gsoc starts but you can use these as proof of work.

DhairyaMajmudar, 2023-12-25:
> as these are subtask of possible task of a possible gsoc project we might not merge till next gsoc starts but you can use these as proof of work.

I understand : )