CircuitVerse/cv-frontend-vue #213: bug: Red color bug when simulator button clicked

upsaurav12, 2023-12-18:
**Describe the bug**
When i clicked simulator button page renders red color bug occuring. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://circuitverse.org/'
2. Click on 'Simulator button' in nav-bar
3. See error

**Expected behavior**
It should display properly when clicked
**Screenshots**

https://github.com/CircuitVerse/cv-frontend-vue/assets/116784047/c5cc73ab-49c9-4564-aec9-baf2585ee1d3

**Device Information [optional]:**
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Additional context**
May be this is from my side

**Are you working on this issue? (Yes/No)

Arnabdaz, 2023-12-22:
the simulator used in production is the old simulator you can open the issue for the same at [CircuitVerse main repo](https://github.com/CircuitVerse/CircuitVerse) and close this issue.

upsaurav12, 2023-12-22:
so i have to open the same issue in main circuitverse repo??

upsaurav12, 2023-12-22:
Closing the issue

closed 2023-12-22 by upsaurav12