CircuitVerse/cv-frontend-vue #217: fix: Content and button are not visible in svg option

upsaurav12, 2023-12-20:
**Describe the bug**
In Simulator for converting the diagram to svg the buttons and contents are not visible like png,jpeg,bmp,gif

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://circuitverse.org/simulator'
2. Click on 'Download as image'
[image]
3. Click on SVG option
4. See error

**Expected behavior**
[image]

**Screenshots**
[image]

**Device Information [optional]:**
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Additional context**
You can check from your side maybe this is from my side 

**Are you working on this issue? (Yes)**

Arnabdaz, 2023-12-22:
This is a main simulator issue and is not present in the vue simualtor.

upsaurav12, 2023-12-22:
where should i raise this issue ??

upsaurav12, 2023-12-22:
closing this issue

closed 2023-12-22 by upsaurav12