CircuitVerse/cv-frontend-vue #225: Inconsistency: Tools' Components Lack Drag Functionality

NayakPenguin, 2023-12-26:
### **Is your feature request related to a problem? Please describe.**
The elements within tools cannot be dragged, creating inconsistency with other components such as Properties and Circuit Elements, which are draggable.

[image]

### **Are you working on this? (Yes/No)**
YES

pankaj-2503, 2024-01-03:
Hey @Arnabdaz  can i work on this issue  i created a pull request on pr #242  but it was already fixed by someone. I don't want any conflict can you please assign this to me.

NayakPenguin, 2024-01-03:
It's clear from my side @Arnabdaz, you are free to assign this issue to anyone :)

Arnabdaz, 2024-01-04:
you can work on this issue @pankaj-2503

pankaj-2503, 2024-01-05:
Hey @Arnabdaz  i am still working on this but couldn't find how to implement draggable feature to links . Give me some more time i would fix this somehow .

Arnabdaz, 2024-01-05:
> Hey @Arnabdaz  i am still working on this but couldn't find how to implement draggable feature to links . Give me some more time i would fix this somehow . 

- No issues you can take your time

Arnabdaz, 2024-01-17:
@pankaj-2503 any updates ?

Chandrachur67, 2024-01-19:
Hey @NayakPenguin @Arnabdaz in the situation mentioned here, the form or the thing that is popping up blocks other activity within screen, right? in that case I don't think that it matters whether it is draggable or not. 

Am I missing something? It would be great if you could explain your idea in little bit more details so that I can think in that direction.
 
Thank you

NayakPenguin, 2024-01-19:
In my view @Chandrachur67, I believe it's not a necessity but rather a valuable enhancement.

Chandrachur67, 2024-01-19:
ok got it. thanks @NayakPenguin

Kartikayy007, 2025-03-10:
Hey @Arnabdaz I think this issue hasn't still been addressed I think I can work on this feature.