CircuitVerse/cv-frontend-vue #241: Hidden "Sign In" button

Priyanshu-su30, 2023-12-31:
**Describe the bug**
In the home page at top right corner the "Sign in" button is hidden.

**Expected behavior**
The "Sign in" button should be displayed properly.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Are you working on this issue? (Yes/No)**
Yes

pankaj-2503, 2023-12-31:
Hey, this could be done so that button is differentiable then links by adding some backgrounds colors .
[image]

Priyanshu-su30, 2024-01-01:
@pankaj-2503 If you add background to the button it will look odd in the navbar.

pankaj-2503, 2024-01-01:
@Priyanshu-su30   i thought  button should look different than  links  .

closed 2024-01-03 by Arnabdaz