CircuitVerse/cv-frontend-vue #253: update issue & feature request template

Arnabdaz, 2024-01-06:
**Is your feature request related to a problem? Please describe.**
Update the issue & feature request template as in the [main repo](https://github.com/CircuitVerse/CircuitVerse/issues/new/choose)

**Describe the solution you'd like**
- keep all the options as in the main repo template 
- In both bug & feature form add a  field for the user to verify if the user faced the issue with / requesting the feature for the vue simulator. (mandatory field)
- In only bug form add a field to ask the user if the user used simulator with or without backend. (optional field)
- In only bug form  add a filed to ensure if the bug is only in dev server or build or in both. (optional field)

**Describe alternatives you've considered**
N/A

**Additional context**
feel free to discuss if you have any other additional fields that can be included which would make the process smoother for devs.

**Are you working on this? (No)**

pankaj-2503, 2024-01-06:
Sure

pankaj-2503, 2024-01-06:
@Arnabdaz  please review this PR.

DhairyaMajmudar, 2024-01-06:
@Arnabdaz shall I update the readme file by adding
1. Community section
2. Code of conduct
3. folder structure 
4. contributors section

pankaj-2503, 2024-01-07:
Hey @Arnabdaz  you can review this updated Pr https://github.com/CircuitVerse/cv-frontend-vue/pull/255

closed 2024-03-25 by Arnabdaz