CircuitVerse/cv-frontend-vue #260: Bug (UI): When ElementsPanel is moved to overlap the Navbar, it becomes locked there and loses the ability to be dragged.

Chandrachur67, 2024-01-16:
**Describe the bug**
The draggable ElementsPanel can be moved to overlap with the Navbar component and if we leave that there then we lose the functionality to move the ElementsPanel further because the Navbar appears on top of the ElementsPanel Component.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to [cv-frontend-vue](https://circuitverse.netlify.app/simulatorvue)
2. Move the ElementsPanel such that it overlaps the Navbar at the top of the screen.
3. Leave it there
4. Now try to move the ElementsPanel component
5. Error: You won't be able to move it.

**Expected behavior**
I can think of two possible solutions.
1. Making the ElementsPanel appear on top of Navbar making it always free to move.
2. Restricting the ElementsPanel to overlap with the Navbar

**Screenshots**

*This is movable*
[image]
*This is not movable*
[image]

**Device Information [optional]:**
- OS: macOS
- Browser Brave
- Version 1.61.109

**Additional context**
I think need to think of one more edge case i.e. when Navbar options are expanded. Considering this case the first possible solution seems better to me.

**Are you working on this issue? (Yes/No)**
Yes

Arnabdaz, 2024-01-16:
@Chandrachur67 you can work on the issue check now the tabsbar must also be having the same issue after merging #223 fix that also ... mostly its a issue of z-index.

Chandrachur67, 2024-01-16:
Thanks for the clarification. The solution is almost ready. I will raise a pr very soon.

29deepanshutyagi, 2024-11-04:
hi , i want to work on this issue , as there is an inactivity on this issue from various months ,so kindly assign me @Arnabdaz

SrisharanVS, 2024-12-10:
@Arnabdaz 
Hey, I have fixed this.. Can you assign me this, I will create a PR soon..

rajsingh112005, 2024-12-21:
@Arnabdaz  
hey, I Have completed my work on this issue plz assign it to me

SrisharanVS, 2024-12-21:
@rajsingh112005 Many did, I think this is ded bro...:)

yaxit24, 2025-01-13:
@Arnabdaz please review my changes [here](https://github.com/CircuitVerse/cv-frontend-vue/pull/404)