CircuitVerse/cv-frontend-vue #263: Feat: Adding design to the SignIn Link.

Sahil-K123, 2024-01-20:
Changing look of SignIn look on hovering over it.

**Additional context**
[image]

Sahil-K123, 2024-01-20:
Hii @Arnabdaz , please assign this issue to me.

subhamkumarr, 2024-01-22:
@Arnabdaz , I want to work on this issue.

anuj-pal27, 2024-01-25:
assign it to me i have already done it 
[image]---->
[image]

anuj-pal27, 2024-01-25:
@Arnabdaz can i make a PR