CircuitVerse/cv-frontend-vue #269: Button Toggle Not Responding

vinayakjaas, 2024-01-26:
**Describe the bug**
The button with the class tabsbar-toggle is not responding as expected when clicked. The button is intended to trigger the toggleHeight function, toggling the display between a chevron-down and chevron-up icon based on the showMaxHeight condition. However, upon clicking the button, there is no observable change in behavior.

**To Reproduce**
Navigate to the page containing the button with the class tabsbar-toggle.
Attempt to click the button.

**Expected behavior**
Upon clicking the button, it should trigger the toggleHeight function and update the icon class accordingly. If showMaxHeight is true, the icon should change to 'fa fa-chevron-down'; otherwise, it should change to 'fa fa-chevron-up'.

**Screenshots**

[CircuitVerse.webm](https://github.com/CircuitVerse/cv-frontend-vue/assets/96948435/981aeeef-87ba-47eb-8986-c1e2dd81afd3)

[image]

**Device Information [optional]:**
- OS: Windows 11
- Browser chrome 
- Version 120.0.6099.227

azlan18, 2024-02-12:
Can I work on this @vinayakjaas ?

Tushargupta002, 2024-12-11:
Is this issue is still open ?
Can i work on this?