CircuitVerse/cv-frontend-vue #375: 🐞 Bug: [SignIn Page is Not Found]

chaitanyakatore, 2024-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues.

### Describe the bug

## Issue Description: 
-  The sign-in button on the  is not functioning as expected. When clicked, it does not redirect the user to the sign-in page, or the sign-in page is not accessible via the expected path. 
- Please investigate and address this issue to ensure proper navigation to the sign-in page.

### Steps to Reproduce

- go to https://circuitverse.netlify.app/simulatorvue
- on this home page there is ** Sign In ** button
- click on it. It will show you the page not found

### Expected Behavior

 - when you click on the Sign In button it must show the sign Page 
 - if user not have account then must give option for the registration

### Screenshots

[image]

### Is the faced issue/bug related to the Vue simulator?

Yes

### Used Vue simulator with or without backend?

Without Backend

### Is the bug present only on the dev server, the build, or both?

Both

### Device Information

```markdown
Os: MacOs
Browser: Safari
```

### Are you working on this issue?

Yes

niladrix719, 2024-08-29:
you have to run the [backend](https://github.com/CircuitVerse/CircuitVerse) on port 3000 first then you will be able to access the sign in page

yashsaraswat2004, 2024-11-24:
can i work on this issue

Arnabdaz, 2025-01-15:
as @niladrix719 said you need to run backend, on netlify we serve only the vue simulator static code without backend. Hence the signIn page not working.

closed 2025-01-15 by Arnabdaz