CircuitVerse/cv-frontend-vue #401: Feature Add a Landing Page to the CircuitVerse Vue Simulator

gitsofaryan, 2025-01-09:
### Is your feature request related to a problem? Please describe

At present, the CircuitVerse Vue Simulator lacks a dedicated entry point or an onboarding page that offers an overview of the application and guides users through its primary features and functionalities. The absence of a landing page creates friction for first-time users who may struggle to comprehend the scope and potential of the simulator. Furthermore, there is no centralized hub to access documentation, tutorials, and user guides, leading to a disjointed user experience.

### Describe the solution you'd like

We propose the development of a landing page that will serve as the primary user interface when accessing the CircuitVerse Vue Simulator. The landing page should be constructed as a single-page application (SPA), leveraging Vue.js for dynamic content rendering and Vue Router for seamless navigation. 

https://github.com/user-attachments/assets/fef182df-2040-4ad3-be31-b9e24964cdcf

### Is requested feature related to vue simulator?

Yes, the landing page is intended to act as the entry point to the CircuitVerse Vue Simulator, and will be tightly coupled with the existing Vue.js application. It will offer contextual information, access to the simulator, and various supporting resources such as documentation and tutorials.

### Are you working on this?

Yes

gitsofaryan, 2025-01-09:
@Arnabdaz @tachyons @tanmoysrt please assign me this work and please me know for changes.

tachyons, 2025-01-09:
Vue simulator will be integrated to main simulator, no separate landing page needed

closed 2025-01-09 by tachyons