CircuitVerse/cv-frontend-vue #406: Feature

nickhil-verma, 2025-01-13:
### Is your feature request related to a problem? Please describe

## Feature Request: Display Element Names or Labels Below Simulator Elements

### Problem Description

The current design of the simulator presents a challenge for users, particularly new users, in understanding circuit elements. While diagrams are helpful, they are not always sufficient for conveying the purpose or function of each element, especially in the miscellaneous section.

 

### Describe the solution you'd like

### Request

I propose that the names or labels of the simulator elements be displayed below them. This enhancement would improve searchability and make the interface more user-friendly, allowing users to quickly identify and understand the circuit elements they are working with.

### Is requested feature related to vue simulator?

#### This problem is related to Simulator 

[image]

### Are you working on this?

Yes

Arnabdaz, 2025-01-15:
we already have tooltips for the elements and the icons are self explanatory
cc @tachyons wdyt

nickhil-verma, 2025-01-19:
@Arnabdaz, I agree that tooltips are self-explanatory, but I don't feel the same for all of them, especially in the miscellaneous section, such as the adder and TB output. I would like to contribute to enhancing them.

Let me know if you'd like further refinements!

Arnabdaz, 2025-01-19:
cc @nitin10s @Waishnav