CircuitVerse/cv-frontend-vue #408: 🐞 Bug: [Title] Circuit Names Not Preserved Individually Across Tabs, Defaulting to Undefined

venkateshkasani, 2025-01-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues.

### Describe the bug

1. When a new circuit is created, the respective circuit-tab shows its name defined by the user. But the circuit name inside the properties panel is not updated and it initially remains empty.
2. Incase of multiple circuit tabs, when a user changes circuit name of any single circuit via the properties panel and switches tab to another circuit, the circuit name inside the properties panel does not update with respective circuit name.

### Steps to Reproduce

1. Go to: https://circuitverse.netlify.app/simulatorvue/
2. Create a new project and open two different circuit tabs.
3. Try to change the name of a circuit using the properties panel.
4. Switch to another circuit tab to see how the same circuit name persists across all the circuit tabs.

### Expected Behavior

1. When user creates a new circuit with a name, the properties panel must reflect the respective circuit name.
2. Individual circuit changes should not mutate the remaining circuit entity names.

### Screenshots

https://github.com/user-attachments/assets/173f48c0-b5c5-4e52-b3bc-41104045db49

### Is the faced issue/bug related to the Vue simulator?

Yes

### Used Vue simulator with or without backend?

Without Backend

### Is the bug present only on the dev server, the build, or both?

Both

### Device Information

```markdown
OS: Windows 10
Browser: Chrome 132.0
```

### Are you working on this issue?

Yes

venkateshkasani, 2025-01-18:
@Arnabdaz Please take a look at this, and could you assign me this issue ?

Avichal296, 2025-01-20:
@venkateshkasani  could we collaborate on this issue

venkateshkasani, 2025-01-20:
It is just a bug and almost solved. You can take up another issue. > @venkateshkasani  could we collaborate on this issue

Avichal296, 2025-01-20:
ok

Arnabdaz, 2025-02-10:
@venkateshkasani I am unable to recreate this issue in our latest build. Are you using the latest version of the code ?

cc @tachyons

closed 2025-02-13 by venkateshkasani