CircuitVerse/cv-frontend-vue #412: 🐞 Bug: eslint not configured for typescript

adityanandanx, 2025-01-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues.

### Describe the bug

[image]

[image]

- Various eslint errors while using typescript

### Steps to Reproduce

1. Setup the project using the [installation guide in the readme](https://github.com/CircuitVerse/cv-frontend-vue?tab=readme-ov-file#installation)
2. temporarily change the lint script in `package.json` to not write any changes - 
    ```
    "lint": "eslint --ext .js,.vue --ignore-path .gitignore --no-fix src",
    ```
3. run `npm run lint`
4. see many problems
[image]

### Expected Behavior

Ideally, no eslint errors. But since this can be a really big change, we can atleast start by letting eslint know that there will be typescript code in the project by using the eslint-typescript plugin.

### Screenshots

[image]

### Is the faced issue/bug related to the Vue simulator?

Yes

### Used Vue simulator with or without backend?

Without Backend

### Is the bug present only on the dev server, the build, or both?

Dev Server Only

### Device Information

```markdown
OS: Arch Linux
Browser: Chrome
Version: 130.0.6723.91 (Official Build) (64-bit)
```

### Additional Context

My first approach to solve this issue would be to add the eslint-typescript plugin to allow eslint to recognize typescript. There's also some `globalVariables` involved so there must be a global declaration file `global.d.ts` which declares the types for the appropriate global variables.   I don't know what is the status of adopting typescript in this project is, but things seem to be all over the place with variables with any declarations and js being used at various places.  This issue may just be redundant as this may be something which goes rather deep, but I can atleast start clearing the top of the rabbithole and do my part.  Thanks

### Are you working on this issue?

Yes

adityanandanx, 2025-01-20:
I would really like a confirmation before I start working on this as I stated before, this issue may just be redundant. 😄
Thanks

niladrix719, 2025-01-21:
Hi @adityanandanx I am assigning you, go for it

adityanandanx, 2025-01-21:
Hey @niladrix719 
I just opened a PR. Let me know if anything seems out of place. I'll be more than happy to go back and forth till our requirements are met. 😄 
Thanks

harikrishna-au, 2025-07-14:
hi , i want to work on this issue , as there is an inactivity on this issue from various months ,so kindly assign me @Arnabdaz

tachyons, 2026-10-13:
Possibly outdated: the repo now lints TypeScript via oxlint (`npm run lint` finds `*.ts` files). Maintainers - does this still reproduce, or can it be closed?

closed 2026-10-13 by tachyons