CircuitVerse/cv-frontend-vue #414: Typescript Integration in `/simulator/src` files

niladrix719, 2025-01-21:
We have already started to migrate the javascript files to typescript, but some are yet to be converted

mainly inside of the` /simulator/src` dir

Pull requests should be made for each files separately feel free to open pull requests

nickhil-verma, 2025-01-21:
@niladrix719 assign me this issue

adityanandanx, 2025-01-22:
Hey @niladrix719 
I think the PR #416 once merged will significantly help identifying ts and eslint errors which will help solve this issue faster.
Our first approach should be to reduce(Ideally remove) the use of `any` for variable declarations rather than just converting js files into ts files where many variables use any as their type.

Let me know your thoughts on this
Thanks

ThatDeparted2061, 2025-01-22:
we can firstly try to avoid using any in the file conversion,
secondly using linting would set us back, for js  and ts file work different in property declarations etc, this wud need us to  first  change the declarations of many properties we were using earlier for js files. That ll take a lot of time.

We already have many declaration issues  to work upon.

JoshVarga, 2025-01-29:
https://github.com/CircuitVerse/cv-frontend-vue/pull/434 closed this issue automatically. Reopening since this is not resolved.

mahmoodalisha, 2025-02-07:
I have created a PR that tries to solve this issue https://github.com/CircuitVerse/cv-frontend-vue/pull/460   
 https://github.com/CircuitVerse/cv-frontend-vue/pull/454 please look into it @vedant-jain03  @tachyons and @ThatDeparted2061

blazethunderstorm, 2025-06-07:
@niladrix719 can i work on this ?

ThatDeparted2061, 2025-06-07:
Yeah go ahead

blazethunderstorm, 2025-06-07:
@ThatDeparted2061 thanks

PR #434 (2025-01-28): refactor : typescript integration src/simulator/src/contention.ts
Fixes #414
@niladrix719 @JoshVarga @Arnabdaz @devartstar 

## Summary by CodeRabbit

- **Refactor**
	- Migrated `ContentionPendingData` class from JavaScript to TypeScript
	- Added type definitions for node management
	- Enhanced contention tracking with more robust methods
	- Improved type safety and error handling for node-related operations
PR #418 (2025-01-28): Js to TS : simulator/src/data/redo.ts 
Fixes #414 
JS to TS
@niladrix719 @JoshVarga @Arnabdaz @devartstar
PR #442 (2025-01-31): refactor: typescript integration : src/simulator/src/wire.ts
Fixes #414 

@JoshVarga @Arnabdaz @niladrix719 @devartstar 

## Summary by CodeRabbit

- **Refactor**
	- Enhanced functionality and type safety for the Wire class.
	- Introduced a new Scope interface to manage wire properties.
	- Updated methods to handle wire coordinates, connections, and interactions, including double-click event handling and visualization.

- **Chores**
	- Improved code clarity without changing core functionality.
PR #423 (2025-02-07): JS to TS simulator/src/hotkey_binder/model/actions.ts
Fixes #414 
Js to TS
@niladrix719 @JoshVarga @Arnabdaz @devartstar 

## Summary by CodeRabbit

## Summary by CodeRabbit

- **New Features**
	- Introduced enhanced key management for user interactions in the simulation.
	- Added functionality for custom key bindings with local storage support.
	- Implemented user feedback for key conflicts to improve user experience.
	- Expanded simulation area manipulation based on user input.
	- Added access to hotkey settings and documentation resources.

- **Refactor**
	- Enhanced type safety for hotkey- [... cut]
closed 2025-06-14 by Arnabdaz