CircuitVerse/cv-frontend-vue #420: 🐞 Bug ( optimzation ) : Testbench driver location

ThatDeparted2061, 2025-01-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues.

### Describe the bug

The location of the testbench driver shouldn't be over other menu options move it below the menu or place it somewhere better.

### Steps to Reproduce

Open the simulator it'll be visible

### Expected Behavior

Shouldn't be hovering over other menu options.

### Screenshots

[image]

### Is the faced issue/bug related to the Vue simulator?

Yes

### Used Vue simulator with or without backend?

Without Backend

### Is the bug present only on the dev server, the build, or both?

Both

### Device Information

```markdown
Ubuntu 24.04 LTS
Chrome 132.0.6834.83
```

### Are you working on this issue?

No

maheshcodes12, 2025-01-24:
Can I work on this issue?

ThatDeparted2061, 2025-01-24:
Yes you can, In your PR do give a reason for your choice for positioning if the option.

092vk, 2025-02-11:
Its draggable , you can position it anywhere , also it overlaps only when one of the menu is open.

ThatDeparted2061, 2025-02-11:
I know but for screens you wud have to readjust it everytume you open anything below outputs, which is easily avoidable this way