CircuitVerse/cv-frontend-vue #433: Removal of JQuery to Vue's reactives

niladrix719, 2025-01-25:
Large part of the JQuery already has been removed, now its the time to remove the rest

Note: Pull requests should be made for each files separately feel free to open pull requests

advait-zx, 2025-01-25:
where's the file/folder

niladrix719, 2025-01-25:
@advait-zx you can search for jQuery `$` in the project using you code editor to find the JQuery code

ThatDeparted2061, 2025-01-26:
@niladrix719 I don't see anymore vue files using jQuery, it's still in use in js files but that wont be replaced by vue's reactivem we'd need to use DOM

harith-hacky03, 2025-01-28:
Hey @niladrix719,

I’ve seen several issues raised related to Vue.js and jQuery, and I believe migrating the website to React/Next.js could be a solid solution. Here are a few advantages React/Next.js offers over Vue.js:

Built-In Server-Side Rendering: Next.js provides native SSR and static site generation, which would improve both performance and SEO.

Larger Ecosystem: React has an extensive ecosystem with many third-party libraries, making it easier to implement advanced features.

Better Scalability: React’s unopinionated nature is great for building scalable and maintainable projects.

Performance Optimization: Features like automatic code splitting in Next.js enhance load times and overall performance.

Stronger Community Support: React has a larger community, offering better resources and support for development.

I’m ready to take on the migration and ensure everything works seamlessly with an improved experience. Let me know if I should proceed!

code-x-r, 2025-01-30:
Hey @niladrix719, please assign these issues to me.

@harith-hacky03 
Vue is not just an alternative but the better-suited choice for CircuitVerse because it:
-> Already fits into the existing tech stack (less rewriting needed).
-> Provides better DX with intuitive syntax and simpler state management.
-> Has Nuxt.js for SSR, making SEO concerns irrelevant.
-> Offers built-in optimizations, reducing dependency bloat.

Instead of shifting to React/Next.js, the focus should be on replacing jQuery with Vue-based solutions to improve consistency and maintainability. This would be easier and faster to do than migrating the whole codebase to react/next.js

niladrix719, 2025-06-04:
@blazethunderstorm yes this issue is open to work for anybody just 1PR per file please

blazethunderstorm, 2025-06-04:
@niladrix719 alright thanks

blazethunderstorm, 2025-06-07:
@niladrix719 i have made some prs pls review them

pantha704, 2026-01-11:
I've submitted a PR for removing jQuery from `src/simulator/src/ux.js`: #761

This migrates all jQuery usage to native DOM APIs as part of the file-by-file approach mentioned in this issue.