CircuitVerse/cv-frontend-vue #468: 🐞 Bug: Fix Circuit generation from the combinational analysis / boolean logic table.

Arnabdaz, 2025-02-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues.

### Describe the bug

Currently creating a circuit from the Combinational analysis / boolean logic table creates a broken circuit.

[image]

### Steps to Reproduce

1. Click Tools > Combinational Analysis
2. Enter a Boolean function (e.g. ABC)
3. click next , it generates the boolean table 
4. Now click generate circuit

### Expected Behavior

The generated circuit should not be broken, Reproduce the same steps in [old simulator](https://circuitverse.org/simulator) to generate a working circuit.

[image]

### Is the faced issue/bug related to the Vue simulator?

Yes

### Used Vue simulator with or without backend?

Without Backend

### Is the bug present only on the dev server, the build, or both?

Both

### Device Information

```markdown
OS: Mac os
Browser: Chrome, firefox
Version: latest
```

### Are you working on this issue?

No

salmoneatenbybear, 2025-02-10:
I would like to take up this issue

salmoneatenbybear, 2025-02-11:
@harikrishna-au 

Would be happy to get help on this issue... I am trying to figure out how to fix this, if you find a solution to this and can get this fixed feel free to open a pr, no issues with me... Let's get this fixed together as quickly as possible and in the best possible way

Arnabdaz, 2025-02-11:
@harikrishna-au you can collaborate with @salmoneatenbybear on this issue

harikrishna-au, 2025-02-12:
@Arnabdaz I fixed the bug but have some doubts. I’d like to ask you something about the code and verify the results. This was quite challenging since it's my first issue.

harikrishna-au, 2025-02-12:
https://github.com/user-attachments/assets/5b72376a-b9d2-49bf-8ebd-0453ea83ec68

@Arnabdaz please go through this

harikrishna-au, 2025-02-13:
@JoshVarga @wallento @devartstar I just fixed this issue. Can I open a PR?

ThatDeparted2061, 2025-02-13:
Yeah go ahead, raise the PR

harikrishna-au, 2025-02-14:
@ThatDeparted2061 I need help raising the pull request! It's my first issue, and I'm struggling to structure the best PR. I'm facing some challenges—could you guide me through the process?

ThatDeparted2061, 2025-02-14:
Message me on slack, i ll help you therr

harikrishna-au, 2025-02-25:
@Arnabdaz, I really need a bit of mentoring to close this issue. Could you take a look?

ThatDeparted2061, 2025-02-25:
What do you need help with ?

harikrishna-au, 2025-05-18:
@ThatDeparted2061  I’ve grasped part of it, but I’m still missing quite a bit. Could you help me fix my first bug?

ThatDeparted2061, 2025-05-18:
Bit caught up rn, i'll try to look into it, t4y going through combinational... types named files

PR #558 (2025-05-31): Fix: Correct boolean table parsing in generateBooleanTableData (#468)
Fixes #468
modified: v0/src/components/DialogBox/CombinationalAnalysis.vue

Describe the changes you have made in this PR –

- This PR addresses the same issue as the previously closed PR [#480 ]
- The previous PR was closed due to some unintentional technical errors.
- Re-implemented the fix cleanly in this new PR.

Code Changes:

- Updated generateBooleanTableData function to correctly parse boolean table output values.
- Now properly categorises table row indices by output value '0', '1', or 'x' for each output name. 
- E [... cut]