CircuitVerse/cv-frontend-vue #469: Chore: Reduce usage of set timout and set interval

Arnabdaz, 2025-02-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues.

### Describe the bug

In the vue js code base there are places where we used set interval or set timeout, we want to replace them with [vue lifecycle hooks](https://vuejs.org/guide/essentials/lifecycle) which would reduce the load on browsers for end users.

### Steps to Reproduce

N/A

### Expected Behavior

The usage of set timeout and set interval should be minimized in the vue files.

### Screenshots

N/A

### Is the faced issue/bug related to the Vue simulator?

Yes

### Used Vue simulator with or without backend?

None

### Is the bug present only on the dev server, the build, or both?

Both

### Device Information

```markdown
N/A
```

### Additional Context

### Please make sure to test the functions you change in every possible way so that no functionalities are broken.

### Are you working on this issue?

No

rxiddhi, 2025-02-10:
Please assign me this issue, I can work on it.

Danish0111, 2025-02-10:
I want to work on it

Arnabdaz, 2025-02-10:
This issue won't be assigned to anyone its a open issue, just put the name of the file you are working on here so that others can know.

rxiddhi, 2025-02-10:
I am working on the following files-
ContextMenu.vue
CombinationalAnalysis.vue
ExportProject.vue 
ApplyThemes.vue

Arnabdaz, 2025-02-10:
Only take one file at time @rxiddhi so that others also have a chance to work as its a good first issue.

rxiddhi, 2025-02-10:
sure @Arnabdaz , i will be working on CombinationalAnalysis.vue

ThatDeparted2061, 2025-02-10:
done, ApplyThemes.vue in #441

Rustix69, 2025-02-10:
@Arnabdaz Done with the Properties Panel.

rxiddhi, 2025-02-10:
@Arnabdaz Hi, I did the required changes, but they got merged with my previous PR. I want to create a separate PR for them. Could you guide me on the best way to do this? #465

Jay2006sawant, 2025-02-13:
Hello  @Arnabdaz Would be working on ExportProject.vue

blazethunderstorm, 2025-05-31:
@Arnabdaz i would be working on contextMenu.vue file

blazethunderstorm, 2025-06-07:
@Arnabdaz  thanks would make a pr for it

Aakansha-Chavan, 2025-06-10:
Hi @Arnabdaz I joined the circuit verse organization 2days ago and exploring it, this is really great that for newcomers issues are being reserved that's the real meaning of collaboration . I am going to setup circuit verse today , so is there any file on which I can work?

pranav718, 2025-07-27:
@Arnabdaz I would like to work on ExportProject.vue file