CircuitVerse/cv-frontend-vue #473: 🐞 Bug: HTML Specifications are not being followed which can lead to Hydration Errors.

Rustix69, 2025-02-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues.

### Describe the bug

The HTML specifications is not being followed in two files which can lead to Hydration Errors and Vue can render the wrong data due to the mismatch.

### Steps to Reproduce

1. Go to CodeBase
2. Run ```npm run dev```
3. The terminal will give Error Warning .
4. And after using for few min its giving Hydration Errors

### Expected Behavior

It should not give errors. And the simulator should run flawlessly.

### Screenshots

[image]

### Is the faced issue/bug related to the Vue simulator?

Yes

### Used Vue simulator with or without backend?

Without Backend

### Is the bug present only on the dev server, the build, or both?

Both

### Device Information

```markdown

```

### Are you working on this issue?

Yes

Arnabdaz, 2025-02-11:
I am assigning the issue to you, but src folder changes need to be merged to v0. Once this is done, you can start working on this issue

Rustix69, 2025-02-14:
@Arnabdaz Hey I have pushed the code we can merge it once you reviewed and the previous merge was done .

closed 2025-05-08 by Rustix69