CircuitVerse/cv-frontend-vue #483: 🐞 Bug: Uncaught TypeError: Cannot read properties of undefined (reading 'newElement')

Rustix69, 2025-02-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues.

### Describe the bug

It suggests that your JavaScript code is trying to access the property newElement of an object that is undefined.

### Steps to Reproduce

1. Go to 'https://circuitverse.netlify.app/simulatorvue'
2. Open the Console in Developer Mode.
3. Now try to take any component from the drawing panel.

### Expected Behavior

The expected behavior is context-dependent—whether newElement is a DOM element, API data, a state property, or something else. 

### Screenshots

[image]

### Is the faced issue/bug related to the Vue simulator?

Yes

### Used Vue simulator with or without backend?

With circuitverse.org

### Is the bug present only on the dev server, the build, or both?

Both

### Device Information

```markdown
OS: Ubuntu 20.04
Browser: Chrome
Version: 89
```

### Are you working on this issue?

Yes

Rustix69, 2025-02-14:
@Arnabdaz You can assign me the issue as i was working in the same file .

harshit078, 2025-03-10:
@Rustix69 Still working on this ?

Rustix69, 2025-03-10:
@harshit078 You can collaborate with me to work on these

closed 2025-05-08 by Rustix69