CircuitVerse/cv-frontend-vue #485: Add Enter key submission for Create New Project dialog

Adityaat2810, 2025-02-16:
### Is your feature request related to a problem? Please describe

Currently, users must explicitly click the Submit button to create a new project after entering the project name. To improve user experience, the form should automatically submit when the user presses Enter in the project name input field.

### Describe the solution you'd like

-When user presses Enter key while focused on the project name input field, the form should automatically submit

[image]

### Is requested feature related to vue simulator?

Expected Behavior:
- When user presses Enter key while focused on the project name input field, the form should automatically submit
- This should have the same effect as clicking the Submit button

Current Behavior:
- Users must explicitly click the Submit button after entering the project name
- Pressing Enter in the project name field has no effect

### Are you working on this?

Yes

Adityaat2810, 2025-02-16:
If this is valid issue please assign to me. 
Thank you : )

Danishfarooq313, 2025-02-17:
Hi, I see that someone has expressed interest in this issue. Is it okay if I also work on it?

ThatDeparted2061, 2025-02-17:
@Adityaat2810 Mention the steps to reproduce the error in the steps !

Adityaat2810, 2025-02-17:
Steps 
1 . Click on + icon to add new circuit on top left .
2 . After entering a name of new circuit , user should be able to create new circuit just by pressing enter and should be out of that dialog by pressing Esc key fro better UX .

Arnabdaz, 2025-02-22:
I think we can work on this feature
@Adityaat2810 assigning to you.

Adityaat2810, 2025-02-22:
@Arnabdaz 
created pr for it .