CircuitVerse/cv-frontend-vue #490: Update Developer Documentation for Vue simulator

Arnabdaz, 2025-02-22:
There has been major changes in the vue simulator code-base after the versioning of the simulator has been introduced, update the developer docs accordingly for better understanding of new contributors.

Arnabdaz, 2025-02-22:
@aryanndwi123 assigning it to you for now, feel free to do the changes yourself or you can assign to other contributors and help them do the changes.

ThatDeparted2061, 2025-02-22:
@aryanndwi123 would like to work on this under you !

Jay2006sawant, 2025-02-23:
@aryanndwi123 Would like to work on this under you!

rxiddhi, 2025-02-27:
Hey @aryanndwi123! I’d be happy to collaborate with you. Let me know if I can join in and help with the changes!

Rustix69, 2025-03-04:
@Arnabdaz @aryanndwi123 Can I work on this issue ? I have worked with this before

blazethunderstorm, 2025-06-04:
@Arnabdaz can i work on this issue

blazethunderstorm, 2025-06-07:
@Arnabdaz @aryanndwi123 is this issue open to work

ThatDeparted2061, 2025-06-07:
Yes it is open, draft the documentation I'll review it.

blazethunderstorm, 2025-06-07:
Sure thanks @ThatDeparted2061