CircuitVerse/cv-frontend-vue #491: Sync the src changes to v0 and v1 versions of the simualtor

Arnabdaz, 2025-02-22:
Vue simulator code-base has two versions as of now `v0` and `v1`, currently the vue simulator is using `v0` for the deployment of the vue simulator build. Currently all the changes are being merged to the `src` folder, but after this change our target is to remove the `src` folder and use the version specific src folders for the adding new changes to the simulator.

Arnabdaz, 2025-02-22:
@aryanndwi123 assigning it to you for now, feel free to do the changes yourself or you can assign to any other contributor and help him do the changes.

ThatDeparted2061, 2025-02-22:
@aryanndwi123 would like to work on this under you !

harikrishna-au, 2025-02-27:
@Arnabdaz could you assign me this issue and check the #497 once

blazethunderstorm, 2025-06-04:
@Arnabdaz can i work on this issue pls

tachyons, 2026-10-13:
Possibly outdated: the repo layout has since changed (`src/`, `v0/`, `v1/` all present).

closed 2026-10-13 by tachyons